HWXLR8/beef-board #18: Add Voltex FW mode

HWXLR8, 2023-12-19:
Been doing some thinking about this one, and it looks like we are going to need a special FW mode for voltex. There are a couple of hurdles:

1. The [encoders used in cryto's voltex controller](https://www.sensatec.co.jp/e/products/catalog.php/293/ST-ELV-24X35B-K_EN.pdf]) use a single 3 pin connector with a variable voltage output. Our design was made with the assumption that controllers will always use encoders with 2 data lines (either in a 2x 3pin format or combined in a single 4pin). As this is not the case, we can re-use one of the 3-pin encoder inputs as an input for these voltex encoders, but we will have to change how the input is handled in FW.
2. The lighting effects run down the sides on the SVRE9, which would also need special FW. Not even sure what type of LEDs this thing uses, but I will experiment and find out.

For now I think it is acceptable to have this a compile-time flag, eventually I would like this to be toggled using a button combo.

Assigning this to myself for now, @supervaka feel free to chime in with your thoughts

HWXLR8, 2024-05-13:
Support for sensatec encoders has been added in V6. We can keep this issue on hold until you have a board in your hands to try out.

ASleepyCat, 2024-05-13:
> Support for sensatec encoders has been added in V6. @ASleepyCat We can keep this issue on hold until you have a board in your hands to try out.

Since I only have an SVSE5, I won't be able to help on RGB LED stuff with the SVRE9 and FAUCETWO.

HWXLR8, 2024-05-13:
> Since I only have an SVSE5, I won't be able to help on RGB LED stuff with the SVRE9 and FAUCETWO.

No problem, LED support is low priority. I will eventually get my hands on a FAUCETWO

PR #80 (2024-06-26): Add support for multiple controllers
Fairly large PR. Closes #18 and #75.

This PR adds basic support for SOUND VOLTEX controllers, and refactors the code base to be structured for multi-device support.
Tested on my SVSE5 and PHOENIXWAN LMT controllers.

* Add support for switching between device types on boot-up
  * Simple one-time combo, set once and it'll remember what controller type it's supposed to be
* Add support for SOUND VOLTEX controllers
  * Add support for AC-style knobs
    * Seems like the way AVRs do ADC doesn't result in crazy jittering like it does on arcin. Thus, no filterin [... cut]
closed 2024-06-26 by HWXLR8