HWXLR8/beef-board #20: Remote USB board with USB-C

ASleepyCat, 2024-01-06:
Instead of requiring two USB cables to power the IO board, develop a remote USB board that only requires one USB input (probably USB-C with PD).

HWXLR8, 2024-01-06:
Thanks for the feature request! The whole idea of of having a remote board was to allow the flexibility to implement such a feature in the future, and it is indeed on the development roadmap, albeit low priority.

ASleepyCat, 2024-06-14:
Apparently a USB-C specific USB 3.x mode can supply up to 3A? https://en.wikipedia.org/wiki/USB-C#USB_3.0/3.1/3.2
This might be enough where USB PD isn't needed.

HWXLR8, 2024-06-14:
Unfortunately the AT90USB1286 is not USB3.0 compliant, so this would not work.

[image]

I think we should still be able to make it work by integrating a USB PD controller on the remote board, but I have yet to do any experiments with this.

ASleepyCat, 2024-06-19:
I managed to power on my PHOENIXWAN with one USB-C cable using this [USB-C breakout board](https://www.amazon.com.au/dp/B0CXS6X4T6?psc=1&ref=ppx_yo2ov_dt_b_product_details). I think we could fairly easily adapt this design onto our daughter board.

It draws just under 2A with all lights on at full bright.
[image]

HWXLR8, 2024-06-21:
I agree, we could easily adapt this, but I only have 1 concern. USB3.0 and 3.1 (even if you're using USB-C) still only has a power cap of 900mA. Your motherboard is just violating those specifications (as many do). Furthermore, the AT90USB1286 is only USB2.0 compliant. It is my understanding that the USB PD specification is entirely separate from the USB2/3/3.1 spec and will allow for a negotiation of higher currents even for USB2.0.

ASleepyCat, 2024-06-21:
Are you sure? On Texas Instruments's USB Type-C and PD PDF, they list USB 2.0 w/ up to 3A/15W as one of the most common uses cases (https://www.ti.com/lit/wp/slyy109b/slyy109b.pdf?ts=1718928612049&ref_url=https%253A%252F%252Fwww.google.com%252F)

I've also read on various threads that USB power specs are actually minimum guarantees and not limits, although I don't know have a primary source for that.

HWXLR8, 2024-06-21:
[image]
Well would you look at that. I had no idea that this was the case. This is fantastic news, thanks for bringing it to my attention.

As for the minimum guarantees, for our use case a "minimum guarantee" is equivalent to a power cap. We do not know who will be plugging the device into what, and if their mobo only supports the minimum gurantee, we will trigger over-current protection. Regardless, if we offer USB-C this is a non-issue. I still wish to offer both choices, and as long as we make it clear in the documentation the limitations of USB-B, this should be no problem.

HWXLR8, 2025-05-02:
I have revisited this today, and I believe that the minimum guarantee of 1.5A would not be sufficient for our use case (your example shows ~2A peak current draw). Not all PC motherboards with USB-C ports will support the higher current modes:

> USB-C ports can advertise higher current capabilities-specifically, 1.5A or 3A at 5V-via the CC (Configuration Channel) pins. However, this is not mandatory for all USB-C ports. Only ports that are specifically designed and configured to provide 3A at 5V will do so; others may only offer the default current.

I wouldn't want someone to plug in a beef board just because its USB-C only to trigger OCP on their motherboard. @ASleepyCat what are your thoughts?

ASleepyCat, 2025-05-02:
> I wouldn't want someone to plug in a beef board just because its USB-C only to trigger OCP on their motherboard. [@ASleepyCat](https://github.com/ASleepyCat) what are your thoughts?

YubiDeck has two USB-C ports where you can plug in both cables if your ports can't supply enough power on one cable. Maybe we can do something similar?

I also think Windows will step in if it detects a USB device drawing too much power, I got that warning when I tried using the remote board with only one cable.

If we're going to stick with using two cables minimum, I think we should replace the USB-B ports with USB-C since I think people have more USB-C cables lying around than USB-B cables.