HWXLR8/beef-board #21: Set up guide?

ASleepyCat, 2024-01-09:
Wiring up the buttons is straightforward, but the turntable is unresponsive when connecting both photo interrupter cables to `DAO_E1P1` and `DAO_E2P1` and the turntable and centre bar LEDs don't turn on when their cables are connected to `TT_LED` and `DAO_BAR` (centre bar lighting isn't implemented so that's understandable).

I'm probably doing something wrong, so could you detail which cables connects where?

ASleepyCat, 2024-01-09:
I changed back to the stock board and everything works fine there, so it's an issue with my board,

ASleepyCat, 2024-01-09:
Nevermind, for some reason I confused the TT headers and everything works now.

closed 2024-01-09 by ASleepyCat

HWXLR8, 2024-01-09:
PM'd on sows