HWXLR8/beef-board #31: Implement missing turntable lighting effects

ASleepyCat, 2024-01-20:
Demo can be seen in [tokaku's review](https://youtu.be/q30dw-_Zi90?t=113&si=M5yStXKrlUd08NM7) of the original PHOENIXWAN, which should be the same across the + and LMT versions of the con.

- [ ] Single colour (user configurable)
- [x] Spin
  * This can be improved to be prettier
- [ ] Colour shift
  * User adjustable shifting speed?
- [ ] Static rainbow
- [ ] Reactive spinning rainbow
- [x] Reactive
  * This is currently a two-tone effect, maybe change it to be more like the stock board? Could also keep the current effect as a different mode
- [ ] Breathing

ASleepyCat, 2024-01-20:
infinitas-arcin and roxy-firmware both use FastLED for their RGB effects, but that requires us to switch over to C++ (at least partially).

I tried to make a PoC, but I ran into weird compilation errors after including FastLED. Could just be an issue with how old my install of `avr-gcc` is, haven't tested it with a recent version yet.

HWXLR8, 2024-01-20:
I am aware of FastLED, but I have never used it myself. I was under the impression that it had a dependency on core Arduino libraries, is that not the case? For the record I don't have a problem with switching to C++.

ASleepyCat, 2024-01-21:
FastLED is available as a library in the Arduino IDE, but there's no hard dependency.

closed 2024-02-21 by ASleepyCat