HWXLR8/beef-board #89: Lunatic Rave 2 now working with beefboard anymore

Mechada2dx, 2024-07-22:
Ever since I switched from purple beefboards to the newer black ones, LR2 won't take input from them anymore, digital turntable still works though. I've tried 2 USB mode instead of 1 USB and even tried running LR2 on different computers but still nothing.

HWXLR8, 2024-07-22:
@ASleepyCat there should be no functional difference between these PCB revisions (v4 vs v6), I believe this is a FW regression.

ASleepyCat, 2024-07-23:
@Mechada2dx Could you see if SDVX mode works? Just want to check a theory.

@HWXLR8 Maybe, if there's a handy way of embedding the current commit hash into the fw. I know roxy-firmware shows the board fw version somewhere in roxy-tool (and probably the same with whatever arcin uses for its config tool), but we only have a rudimentary fw flashing tool.

ASleepyCat, 2024-08-03:
I've identified the issue and pushed the fix as part of #88. It's untested though since I can't get to the key config, but button inputs should at least be working again.

closed 2024-08-19 by HWXLR8