HWXLR8/beef-board #90: Embed FW version into FW

HWXLR8, 2024-07-23:
We currently have no idea what FW version is flashed to the board. For debugging purposes, this is crucial information. My proposed solution is to write the FW version to a specific address in EEPROM, then read back the value in beef-tool using `avrdude`. My main concern is what constitutes the "FW version". Since we combine the pcb design, fw source, and beef-tool all in one repo, changing things completely unrelated to the FW will change the hash. We could introduce our own versioning system, but that feels a bit silly. @ASleepyCat thoughts?

ASleepyCat, 2024-07-23:
I guess the "proper" solution would be to split the hardware, firmware, and utility tools into their separate repos, akin to what `roxy-board` does. `beef-board` is essentially a monorepo, and there's no good way to do separate versioning of individual components AFAIK for that style of repo (at least without a manual versioning system).

We could just be lazy, and take the commit hash and say that's good enough since we'd be able to look through commit history to see any nearby relevant changes to the problem commit.

HWXLR8, 2024-07-23:
I elect that we take the lazy approach since the hardware components going to be receiving very infrequent updates going forward. I'll take ownership of this issue since I worked on the FW flashing tool.

ASleepyCat, 2024-07-24:
I think ideally we'd use HID to report back this type of information, but yeah storing it in EEPROM will do for now.

closed 2024-09-26 by HWXLR8