HWXLR8/beef-board #95: PS2 support

ASleepyCat, 2024-08-28:
I guess this is something someone might want in the future.

Might require a hardware update. [roxy-firmware uses SPI](https://github.com/veroxzik/roxy-firmware/blob/roxy/roxy/spi_ps.h) for PS2 support, not sure how that would work in AVR.

HWXLR8, 2024-08-28:
So this connector breaks out the SPI pins on our microcontroller:

[image]

we would need to make a special adapter which plugs into this, but at least no new hardware revision is required

ASleepyCat, 2024-08-29:
What about Dao's [PS2 adaptor](https://www.gamo2.com/en/index.php?dispatch=products.view&product_id=364)?

HWXLR8, 2025-04-19:
> What about Dao's [PS2 adaptor](https://www.gamo2.com/en/index.php?dispatch=products.view&product_id=364)?

I wonder if this is doing any magic handshake with DAO controllers, or whether it just translates HID. I need to get my hands on one of these.

ASleepyCat, 2025-04-20:
> > What about Dao's [PS2 adaptor](https://www.gamo2.com/en/index.php?dispatch=products.view&product_id=364)?
> 
> I wonder if this is doing any magic handshake with DAO controllers, or whether it just translates HID. I need to get my hands on one of these.

Probably (hopefully) no weird magic/proprietary protocols, considering it supports official Konami controllers.

HWXLR8, 2025-04-20:
Managed to get my hands on an adapter and a PS2 copy of Empress. Should be arriving in the next few weeks, will update with results