HWXLR8/beef-board #96: Implement custom bootloader

ASleepyCat, 2024-08-28:
Would allow us to automatically switch beef-board into bootloader mode and back to application code without having the unplug and replug the board.

Also serves as groundwork for flashing via a config tool in the future.

Some references:
* [LUFA's HID bootloader](https://github.com/abcminiuser/lufa/tree/master/Bootloaders/HID)
  * Only the bootloader portion, doesn't have examples for how to switch into bootloader mode from application code
* [mon's Pocket Voltex](https://github.com/mon/PocketVoltex)

HWXLR8, 2024-08-28:
So the way Arduino bootloader decides whether to jump to bootloader is to listen for a sequence like:

1. Host connects to mcu via serial terminal
2. Host requests change in baud rate

This is a strange enough sequence that the user would never do it, and it still allows to the user to use a serial terminal if they need it.

We do not have to have as complicated of a handshake as the arduino bootloader since we are not a generic microcontroller pcb, maybe just listen for a serial terminal connection and jump to bootloader when the connection is established.

ASleepyCat, 2024-08-29:
I was more thinking of implementing an HID bootloader so that our board doesn't require drivers to update fw. But I imagine we won't be able to feasibly flash a custom bootloader onto existing boards, so should we just use the default bootloader that comes with our MCU?

ASleepyCat, 2024-09-01:
I'll work on this after #98. Our detection method might need to change depending on what we're able to do from the browser.