HWXLR8/beef-board #112: Web Config site unable to function due to WebUSB blocking

drugexchange, 2024-10-13:
There is currently an issue where me and another user have an issue attempting to use the web config provided on the readme. Upon attempting to connect the board to the webconfig any chromium browser cannot detect it. I took a look through chrome://usb-internals and it did report that WebUSB was blocking it from interacting with anything on chrome. Unsure if this is the reason that the board cannot be used on the web-config.

HWXLR8, 2024-10-13:
@ASleepyCat Thoughts?

ASleepyCat, 2024-10-14:
I'm assuming you're using Chrome? Chrome should support WebHID, I tested with Edge and that also works. Have you changed some privacy settings that might block WebHID?

drugexchange, 2024-10-14:
Checked my privacy settings HID isn't getting blocked by it. However, I am using the stock firmware that was used. Unsure if its a recent firmware or outdated one.

HWXLR8, 2024-10-14:
@ASleepyCat The firmware I shipped should have been from this commit 4b8f684e9ef40fe88813fcf7988640c37cfeebc6. It should be pretty recent.

@drugexchange could you please try updating to the latest FW anyway? You can find instructions here:

https://github.com/HWXLR8/beef-board/tree/master/beef-tool

You can manually put the pcb into programming mode by pressing the button in the corner of the board. You can find both the latest FW and the FW flashing utility here:

https://github.com/HWXLR8/beef-board/actions/runs/11274018346

drugexchange, 2024-10-14:
Great news for both of you guys after flashing the firmware provided by @HWXLR8 it now gets detected by the web config so yeah probably the wrong firmware got put on the board. This can now be closed by either of you two.

closed 2024-10-14 by ASleepyCat