HWXLR8/beef-board #137: Board seems to shut off on its own after a few minutes of playing (9keys/popn)

abadejos, 2025-07-20:
Hello, one of my friends gave me a beef board to drop into my pop'n dao. However, sometimes when I'm playing, the board will seemingly "shut off", no longer registering any inputs, and it doesn't turn back on again. Even unplugging it and plugging it back in doesn't seem to work. The only solution is to restart my computer, and even then, sometimes it's not detected by my computer.

My friend flashed it with the newest firmware on github, but the webui seems to say that my firmware is out-of-date (if it's detected).
We don't think it's the cable, because he tested it on another device for a good while, and it worked fine. I also wiggled the board to see if there was any loose connections, it didn't really make a difference. I've also disabled LEDs in the webui, it doesn't seem to change anything. The controller is currently on joystick mode.

I've tested this on: pop'n data (through spice2x and on windows), Lunatic Rave 2 (on windows), and Beatoraja (on Linux). Initially, I thought this was an issue with Linux, but it seems to do this strange shutting-off even on windows. If there's anything I can try to fix this issue, I'd greatly appreciate it.

HWXLR8, 2025-07-20:
Sorry to hear you are having issues. A couple of questions:

1. You mentioned that you tried on Linux/Windows, are these both on the same PC? Is this behavior observed on multiple PCs? Just want to rule out the host.
2. Is the PCB being used in 1-USB or 2-USB mode? You can check the jumper in the bottom right:
  [image]

3. Was this board assembled by your friend? Or was it purchased through beef on discord?

@ASleepyCat What is the FW version cutoff for the web UI flashing? Theoretically they should be able to update to latest using beef-tool then switch to web UI afterwards, right? Although I do not suspect the FW since they mentioned that unplugging/replugging has no effect.

abadejos, 2025-07-20:
hi, i'm going to answer what i can and edit later, it's like 4 am in my timezone so i cant open the controller right now. i'll update this when i'm able to, thank you both for the prompt responses

1. yes, this is on the same pc. i will add that he gave me an arcin board earlier, and it had the same issues (board shutting off). we thought it was a faulty board so he gave me a beef board
2. will check in a few hours, can't open up controller right now
3. no, it's not recognized by the os. i've been plugging it in and unplugging it on the front panel (i've tried the back and it doesn't seem to make a difference), and it's not being recognized right now at all. can't check the kernel right now since it's been a few days that i've played a game with this on linux. i do have this in my dmesg though... not sure what over-current condition is:

[image]
(The latest message popped up when I unplugged the beef board from my computer)
(I disabled XHCI on my BIOS for right now, and I set legacy USB support to auto. That's all I changed in my bios settings. )
4. this board was bought through someone on discord. he did not assemble it
5. can't check hash display because the board isn't being recognized at all at the moment

I'm noticing odd behavior when I plug/unplug the controller. It's not a device on lsusb, yet, when I plug it in, my phone no longer gets recognized as a storage device and a few devices disappear. First one is with the controller unplugged, and the second one is when it's plugged in. 

[image]

HWXLR8, 2025-07-21:
A couple of things stand out to here. 

1. The issue was present with the arcin, before beef board was introduced into the system.
2. The over-current protection (OCP) is being triggered on _multiple_ ports.

OCP is a protection mechanism which prevents your USB ports from drawing too much current, and it will cut off current to the device to protect it. This is why you are seeing the "shutting off" behaviour. OCP often takes some time to cool down once it is triggered, which explains why sometimes even rebooting your computer doesn't help. OCP is not perfect, and it does not guarantee that your motherboard was not damaged by the device. In fact, the motherboard might not have been damaged by a USB device at all. I have seen instances in the past where a general motherboard faults causes OCP and some ports just stop working permanently. Regardless, the fact that this issue was present with the arcin (before beef board was introduced into the system) rules out it being the cause.

Try unplugging your PC from the wall and holding down the power button for a bit, this should drain any residual charge. Once that has been done, plug in a known good device with a known good cable (maybe your smartphone), and check if OCP is still triggered. If it still has, your motherboard is likely damaged.

Closing this issue as I do not believe its an issue with beef board specifically, but feel free to comment.

closed 2025-07-21 by HWXLR8

abadejos, 2025-07-21:
Thanks for the insight... yeah, it seems like I have to replace my motherboard. Oops. Does the beef board run with a lot of current by default, though? I tried disabling LEDs on the webui to see if that'd make it use less power, but it didn't really make a difference. I'm kind of glad my board isn't faulty, though, but I'm just trying to see what I should do so OCP won't trigger in the future.