HWXLR8/beef-board #139: V6 PCB will connect, but not actually register inputs/show lighting.

alixdotsh, 2025-08-20:
Heyo. I’ve been switching my deck from bi2x to beef boards and I’ve run into an issue. Up to just a while ago the board was working perfectly fine, inputs were being registered when pressed, along with their LEDs, turntable lighting was working too, but after more troubleshooting it now does not. I unplugged everything from the board when I was routing my cables in the deck to the board and I plugged everything back in. I tried plugging it into my computer and no inputs or even the turntable are being registered on hardware tester, button LEDs/TT LEDs do not work either. I am on the latest firmware via beef-tool.

Here is a list of things I have tried to do while troubleshooting:
- different data cables
- restarting my pc
- try plugging it into a different pc
- reinstalling drivers and re-flashing firmware
- reset button on pcb
- different ports in computer
- try a different board (I was shipped 2)

If there are more things you’d like to know, or any other methods to troubleshoot, please do let me know and I’ll get back to it as soon as possible.

alixdotsh, 2025-08-22:
The controller is an actual arcade controller yes. A friend of mine was using it for a while before shipping them out to me, they were working when I was plugging everything in initially to double check.
> * I assume your beef board was working before you swapped out your bi2x board, was it working afterwards as well?

Yes. Until I took everything off the board to not have a messy internal when I screw everything back together.
> * Have you tried using your bi2x board again and see if that still works?
The bi2x board still works.
> * Do you have enough slack with your wires after you re-routed them and made sure they aren't damaged in some way?
Define "slack," im still new to this kind of stuff, i also didn't do the wire making myself. I haven't actually checked to see if they are damaged or not properly touching the connectors, but I can try to do that. I will mention that one button LED lit up for a millisecond once when I was reflashing firmware.
> * How many USB cables are you using to power the board?
2 cables. Data, and then power.

I will mention I am running modified firmware, but I highly doubt that is the problem because I used official firmware and it wasn't a fix. The only modification is the amount of LEDs being addressed for the turntable.

HWXLR8, 2025-08-24:
It is pretty difficult to debug HW issues without being physically present. The failure mode is quite strange considering it once worked then suddenly stopped. That rules out many issues with pinouts.

>Up to just a while ago the board was working perfectly fine, inputs were being registered when pressed, along with their LEDs, turntable lighting was working too, but after more troubleshooting it now does not.

Can you elaborate on what occurred during this period?

>2 cables. Data, and then power.

Are both cables plugged into the PC? Both cables supply power to different parts of the circuit so they need to share the same GND reference.

>I will mention I am running modified firmware

Did problems begin after flashing the modified FW or before?

alixdotsh, 2025-08-24:
> Can you elaborate on what occurred during this period? What did you change between when it was working and when it began acting up.

Wednesday morning at id say around 9:30am I had unplugged the board from the PC, and all the cables that were connected to the board. All that I had done was try to cable manage and make things look neater, because admittedly the provider sent me their wires and it wasn't the neatest job, but it worked. I was routing cables through the deck so that there were not just a big group of cables possibly getting tangled within each other. I then started to plug everything back in to the board (id say this is around 10am). Once everything was plugged back in, that is when the issue started.

> Are both cables plugged into the PC? Both cables supply power to different parts of the circuit so they need to share the same GND reference.

Both are plugged in to the PC, I tried just the data cable (where just the button LEDs would work), and then both. Neither was a fix to the issue. It's very weird, because I can control the turntable lighting through the web config, and it works instantly, but nothing else will. Keyboard mode doesn't fix it either.

> Did problems begin after flashing the modified FW or before?

It happened after. But again, I did try to flash back to official FW to see if it would fix it and it did not. The fork of the repository is public and it's on my profile, you can see all that I changed (a single line in the makefile) and the workflows/prod that was being done.

> It is also very telling that the same issue is occurring with both boards, which rules out manufacturing defects. Did the person who provided you with the boards have any issues?

They told me they had no issues as it was working completely fine in their deck a couple of weeks ago, when they were playing BMS. They also have zero idea what is the issue.

Ultimately im just worried that I somehow bricked the boards in a way, as I know that isn't something that can really be fixed. But im trying to be optimistic and not think that's the case.

HWXLR8, 2025-08-24:
Considering you are able to flash the device, the likelihood of the board being bricked is vanishingly small. That being said, there is still something strange going on here. When you changed the firmware, how did you build it? Manually? or using the binary built by the github workflow? How did you flash the firmware? I looked at your fork and I agree that your changes are unlikely to be the cause..

@ASleepyCat do you know of any way in windows to check if the HID device is presenting any buttons?

alixdotsh, 2025-08-24:
> When you changed the firmware, how did you build it? Manually? or using the binary built by the github workflow? How did you flash the firmware?

I built it using the binary built by the github workflow. I don't know if that was exactly the one used in this repository but I tried to match action steps and went from there. I flashed it using beef-tool, as the web config flashing never was a success for me.

alixdotsh, 2025-08-29:
hiya, just giving an update. i left all but one battery plugged in and that was not registering either. sorry for late resopnse, work has been hectic this week.

HWXLR8, 2025-08-31:
Unfortunately at this point there is no other support I can offer. The fact that the board was working before the cable management suggests that some sort of user error has occurred. I will be closing this issue, if in the future you come across any promising leads, feel free to re-open the issue and I will try my best to help.

closed 2025-08-31 by HWXLR8